JetBrains/intellij-micropython #247: Micropython REPL not working

zerovijay, 2023-08-24:
After Pycharm2023.2.1 update REPL still not working

anglerfish27, 2023-09-02:
Same here for Raspberry Pico. In fact its never worked right. I think once I got it working for a short time and then it died. The whole implementation of Micropython is terrible. You really need to support it fully, multiple boards ect... Pycharm is for me at least, the BEST IDE for python/micropython. However its rendered useless with the severe limitations placed on MP for example on flashing main.py. No area to view all your files and open/edit them such as with thonny, and the death nail which is clearly an issue that has NEVER been fully resolved, REPL does not work.

I can get it to push my code to the Pico. Then I have to use another terminal/serial program to view its output. This is not what an awesome IDE should require. I mean Thonny is pretty darn basic, and they got it hammered down solid. Surely the team for Pycharm can do the same.   Sadly due to these limitations I use Pycharm mostly for learning python3. Not MP. I use a mix of Thonny or most of the just VSCODE. 

Fix it already!! This issue stems back years!!

I'm also on PyCharm 2023.2.1 (Community Edition) on Mac OS X (latest version). Really such your syntax and auto fill feature are so great. Really should support MP properly. This can't be a new complaint. Missing out on a whole group of people who would likely pay for Pycharm if MP worked right with all the modern boards.

I see that on the official support side of your website you decline this issue and push it off to here and others. Come on...we want to use your product!!

abayerle1, 2023-10-05:
The REPL just blinks and does nothing in terms of print("hello world") <br> </br>
[image]
<br> </br>
The Arduino serial monitor shows me that it works:
<br> </br>
[image]

zerovijay, 2023-10-05:
I think we lost good ide for micropython hobby projects but we can still use pycharm using minicom serial comments but it's pain in the @#£_& please fix this issue please. last 2 updates I am waiting for fix this but I got nothing i know we have other options like thonny or vs code but I love the futures from pycharm

gabemorris12, 2023-10-05:
I reached out to JetBrains and asked them about the abandoning of this project. They responded and mentioned that a the lead developer on this is no longer working for JetBrains, and they said that they will get around to fixing it. In the meantime, I am still going to use the last working version because it's still very worthwhile to use this over the others in my opinion. It's not so big of a deal if you are using the JetBrains toolbox because it makes it easy to manage your installations of their products.

lancelote, 2023-10-16:
REPL is fixed by #253. I am sorry it took so long.

closed 2023-10-16 by lancelote