JetBrains/intellij-micropython #248: REPL not opening

MRIIOT, 2023-08-29:
```
2023-08-29 13:13:20,973 [  84327] SEVERE - #c.i.o.o.e.ConfigurableWrapper - No display name specified in plugin descriptor XML file for configurable com.jetbrains.micropython.settings.MicroPythonProjectConfigurable;
specify it using 'displayName' or 'key' attribute to avoid necessity to load the configurable class when Settings dialog is opened [Plugin: intellij-micropython]
com.intellij.diagnostic.PluginException: No display name specified in plugin descriptor XML file for configurable com.jetbrains.micropython.settings.MicroPythonProjectConfigurable;
specify it using 'displayName' or 'key' attribute to avoid necessity to load the configurable class when Settings dialog is opened [Plugin: intellij-micropython]
	at com.intellij.openapi.options.ex.ConfigurableWrapper.getDisplayName(ConfigurableWrapper.java:156)
	at com.intellij.openapi.options.ex.Weighted.lambda$static$0(Weighted.java:26)
	at java.base/java.util.TimSort.binarySort(TimSort.java:296)
	at java.base/java.util.TimSort.sort(TimSort.java:221)
	at java.base/java.util.Arrays.sort(Arrays.java:1307)
	at java.base/java.util.ArrayList.sort(ArrayList.java:1721)
	at com.intellij.openapi.options.ex.SortedConfigurableGroup.buildConfigurables(SortedConfigurableGroup.java:41)
	at com.intellij.openapi.options.SearchableConfigurable$Parent$Abstract.getConfigurables(SearchableConfigurable.java:75)
	at com.intellij.openapi.options.ex.EpBasedConfigurableGroupKt.collect(EpBasedConfigurableGroup.kt:148)
	at com.intellij.openapi.options.ex.EpBasedConfigurableGroupKt.access$collect(EpBasedConfigurableGroup.kt:1)
	at com.intellij.openapi.options.ex.EpBasedConfigurableGroup.<init>(EpBasedConfigurableGroup.kt:110)
	at com.intellij.openapi.options.ex.ConfigurableExtensionPointUtil.getConfigurableGroup(ConfigurableExtensionPointUtil.java:119)
	at com.intellij.ide.actions.ShowSettingsUtilImpl$Companion.getConfigurableGroups(ShowSettingsUtilImpl.kt:48)
	at com.intellij.ide.actions.ShowSettingsUtilImpl.getConfigurableGroups(ShowSettingsUtilImpl.kt)
	at com.intellij.ide.actions.ShowSettingsAction.perform(ShowSettingsAction.java:60)
	at com.intellij.ui.mac.MacOSApplicationProvider$Worker.initMacApplication$lambda$2$lambda$1(MacOSApplicationProvider.kt:71)
	at com.intellij.ui.mac.MacOSApplicationProvider$Worker.submit$lambda$6(MacOSApplicationProvider.kt:170)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:208)
	at com.intellij.openapi.application.TransactionGuardImpl.access$100(TransactionGuardImpl.java:21)
	at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:190)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:861)
	at com.intellij.openapi.application.impl.ApplicationImpl$4.run(ApplicationImpl.java:478)
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:79)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:121)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:41)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:792)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:739)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:733)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:761)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.kt:685)
	at com.intellij.ide.IdeEventQueue._dispatchEvent$lambda$10(IdeEventQueue.kt:589)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWithoutImplicitRead(ApplicationImpl.java:1485)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.kt:589)
	at com.intellij.ide.IdeEventQueue.access$_dispatchEvent(IdeEventQueue.kt:67)
	at com.intellij.ide.IdeEventQueue$dispatchEvent$processEventRunnable$1$1$1.compute(IdeEventQueue.kt:369)
	at com.intellij.ide.IdeEventQueue$dispatchEvent$processEventRunnable$1$1$1.compute(IdeEventQueue.kt:368)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:787)
	at com.intellij.ide.IdeEventQueue$dispatchEvent$processEventRunnable$1$1.invoke(IdeEventQueue.kt:368)
	at com.intellij.ide.IdeEventQueue$dispatchEvent$processEventRunnable$1$1.invoke(IdeEventQueue.kt:363)
	at com.intellij.ide.IdeEventQueueKt.performActivity$lambda$1(IdeEventQueue.kt:992)
	at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:105)
	at com.intellij.ide.IdeEventQueueKt.performActivity(IdeEventQueue.kt:992)
	at com.intellij.ide.IdeEventQueue.dispatchEvent$lambda$7(IdeEventQueue.kt:363)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:861)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.kt:405)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:207)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:128)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:117)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:105)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:92)
2023-08-29 13:13:20,976 [  84330] SEVERE - #c.i.o.o.e.ConfigurableWrapper - PyCharm 2023.2.1  Build #PY-232.9559.58
2023-08-29 13:13:20,976 [  84330] SEVERE - #c.i.o.o.e.ConfigurableWrapper - JDK: 17.0.8; VM: OpenJDK 64-Bit Server VM; Vendor: JetBrains s.r.o.
2023-08-29 13:13:20,976 [  84330] SEVERE - #c.i.o.o.e.ConfigurableWrapper - OS: Mac OS X
2023-08-29 13:13:20,976 [  84330] SEVERE - #c.i.o.o.e.ConfigurableWrapper - Plugin to blame: MicroPython version: 1.4.1-2023.2
```

adrianblakey, 2023-09-04:
Please could someone fix this?

gabemorris12, 2023-09-04:
@adrianblakey I am concerned that this project is dead. I've just emailed JetBrains, hoping to get some more information. In the meantime, micropython still works on PyCharm 2023.1.4. See issue #242.

callmerough, 2023-09-09:
In the mean time I got things working via https://github.com/AmirHmZz/mpbridge tool. I've added some shell scripts like this:

For development mode (and REPL) :
`mpbridge dev  --auto-reset=hard --no-prompt c3 . `

For syncing files:
`mpbridge sync -c c3 . `

lancelote, 2023-10-16:
Fixed by #253