JetBrains/intellij-micropython #249: ESP32 - Uploading files: 0% (0/1) - can't flash

Amundw99, 2023-08-30:
Try to Run "flash main.py", but result is Uploading files: 0% (0/1)
Have installed MicroPython plug-in.

Please advice 

PyCharm 2022.3.2 (Pro Ed.)

Hereux, 2023-08-30:
It's the same for me

Hereux, 2023-08-30:
I got it working. It was not a problem of the plugin. When I flashed my ESP with micropython, I used the standart baudrate from the tutorial. In an other tutorial 115200 baud was used. This did not corrupt my firmware and worked perfectly.

https://iludis.de/wp-content/uploads/2019/02/MicropythonWemosD1mini.pdf

Amundw99, 2023-09-03:
Thanks Hereux

I follow your advice up to Page 3 : LED blinken lassen".

Te, Run Flash main.py but nothing happens unless Uploading files: 0% (0/1) . 

Is there possible to set baud rate in Pycharm? Could it be that?

Hereux, 2023-09-14:
Sorry for the late reply, I forgot.

Have you installed the CH340 USB driver? Maybe try reinstalling it, I had a problem with that too.
https://sparks.gogo.co.nz/ch340.html

I can't say exactly what your problem is. Have you tried connecting to MicroREPL?
The plugin repl doesn't work for me. You can use CoolTerm or uPyCraft to connect to the ESP.
I would suggest using CoolTerm:
https://freeware.the-meiers.org

If MicroPython is flashed correctly, it should look like this:
[image]

Make sure you are using the correct port and baud rate. If you want to see what the gibberish above means, you need to connect at a baud rate of 72800. You just get startup information, which you don't normally need, but if Micropython encounters an error, you will see it.
[image]

If it is not flashed correctly, “MicroPython v1.20.0 on 04/14/2023; ESP module with ESP32” not displayed.
In this case, try flashing Micropython again using the following command:
https://micropython.org/download/?port=esp32

esptool.py --chip esp32 --port YOUR_PORT --baud 460800 write_flash -z 0x1000 --flash_size=detect 0 FIRMWARE_NAME.bin
esptool.py --chip esp32 --port YOUR_PORT write_flash -z 0x1000 FIRMWARE_NAME.bin

Unfortunately I only have ESP2866, so I don't know for sure if the above works.

Amundw99, 2023-09-26:
@Hereux thank you, I've been offline, soory for late respond.
I will try testing more of this.

USB driver: I'm using Silicon Labs CP210x USB to UART brigde
FW : from this page https://micropython.org/download/ESP32_GENERIC_S3/ I am using this one https://micropython.org/resources/firmware/ESP32_GENERIC_S3-SPIRAM_OCT-20230426-v1.20.0.uf2

When logging on by Putty, I got this?
[image]

Do You understand the error messages?

Hereux, 2023-10-25:
Sorry, I was offline too. I don't fully understand the error, but I would say that the driver sent something to the board but couldn't write it to flash memory. The OSError says that the flash chip wasn't even initialized. To me it seems like your ESP memory is dead. have you tried a new esp? By the way, I no longer have any errors, which is because I took a new one.

closed 2024-08-29 by elmot

LuFeiX, 2024-12-19:
I also met this problem, Connecting to COM8

Uploading files: 0% (0/1)

C:\Users\18036\PycharmProjects\pythonProject2\light.py -> light.py has been stuck here I am the esp01s firmware download error? In addition [micropython] (https://github.com/JetBrains/intellij-micropython does not support the latest version of the kernel 243 242 can be used at present.