JetBrains/intellij-micropython #270: No 1.4.2 for 2023.3+

emcek, 2023-12-13:
`1.4.2` is not listed for 2023.3 branch
[image]

zerovijay, 2023-12-13:
i can't understand!

emcek, 2023-12-13:
`1.4.2` is compatible only with **2023.1.*** and **2023.2.*** (see table strict ranges)
but for **2023.3.*** is only available `1.4.0` since **2021.3+** regular expression

zerovijay, 2023-12-14:
It's automatically upgraded i can't go below!

zerovijay, 2023-12-14:
If I do something the no longer work micropython depending failed to download
Including manual installation

gabemorris12, 2023-12-16:
@emcek The REPL console still doesn't work on Pycharm 2023.3.1 with Micropython 1.4.0. @zerovijay You can install previous versions of the Micropython plugin through [this page](https://plugins.jetbrains.com/plugin/9777-micropython/versions/stable). Follow the instructions after downloading.

emcek, 2023-12-17:
As compatibly chart show... only `1.4.0` is compatibility with `2023.3.*` branch....
[image]
[image]
[image]

closed 2024-07-13 by lancelote