JetBrains/intellij-micropython #271: MicroPython REPL stil not fixed!

zerovijay, 2023-12-13:
# MicroPython REPL Not Working!

[image]

PyCharm 2023.3.1 (Community Edition)
Build #PC-233.11799.298, built on December 12, 2023
Runtime version: 17.0.9+7-b1087.7 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Linux 6.6.4-200.fc39.x86_64
GC: G1 Young Generation, G1 Old Generation
Memory: 2048M
Cores: 2
Registry:
  ide.experimental.ui=true
Non-Bundled Plugins:
  intellij-micropython (1.4.0)
Current Desktop: GNOME

osarhan, 2023-12-16:
Same here:

PyCharm 2023.3.1 (Professional Edition)
Build #PY-233.11799.298, built on December 11, 2023
Runtime version: 17.0.9+7-b1087.7 x86_64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
macOS 14.1.2
GC: G1 Young Generation, G1 Old Generation
Memory: 2048M
Cores: 8
Metal Rendering is ON
Registry:
  ide.experimental.ui=true
Non-Bundled Plugins:
  intellij-micropython (1.4.0)

lenineto, 2024-07-10:
It's being 7 months now, 100s of complaints about the same issue, and it's still not fixed.

closed 2024-07-13 by lancelote