JetBrains/intellij-micropython #272: Unable to use MicroPython modules in PyCharm

somniumLux, 2023-12-16:
Are there any limitations with MicroPython for PyCharm and other IDEs? Python scripts that work for my RPI Pico board work in Thonny but not in Pycharm and VS Code. Do certain modules have limitations or MicroPython in general because it is configured as by instructions.

Thanks :)

zerovijay, 2023-12-18:
what kind issue you are facing ?

somniumLux, 2023-12-19:
A code example is this:
`from machine import Pin
import time

led = Pin("LED", Pin.OUT)

i = True
while i:
    led.on()
    time.sleep(1)
    led.off()
    time.sleep(1)
`

This code works fine in Thonny but in PyCharm I get this error message:

Connecting to COM6
Traceback (most recent call last):
  File "C:\Users\Lovre\AppData\Roaming\JetBrains\PyCharm2023.3\plugins\intellij-micropython/scripts/microupload.py", line 139, in <module>
    main(sys.argv[1:])
  File "C:\Users\Lovre\AppData\Roaming\JetBrains\PyCharm2023.3\plugins\intellij-micropython/scripts/microupload.py", line 56, in main
    board = Pyboard(port)
            ^^^^^^^^^^^^^
  File "C:\Users\Lovre\PycharmProjects\rpiProject\.venv\Lib\site-packages\ampy\pyboard.py", line 147, in __init__
    raise PyboardError('failed to access ' + device)
ampy.pyboard.PyboardError: failed to access COM6

This is better than it was before because before, it was complaining that it could not find the machine module but now it can't reach the board. In this case, the code was flashed to the board from Thonny successfully and without changing the port, it fails to do the same thing with the same code here in PyCharm.

However, sometimes I get this output but the on board LED is not flashing:

Connecting to COM6
Uploading files: 0% (0/1)
C:\Users\Lovre\PycharmProjects\rpiProject\led_blink.py -> led_blink.py
Uploading files: 100% (1/1)
Soft reboot

Thanks for help in advance :)

gabemorris12, 2023-12-19:
Whenever you're flashing the script using Pycharm, you have to make sure that no other application is connected. It appears as though you are attempting to flash while Thonny is connected. In Thonny, you can disconnect by going to the Run tab and clicking disconnect, then flash using PyCharm. Right now though, I believe that flashing the script is all that can be done. I haven't been able to get the REPL console to work regardless of the version matching as seen in #270.

somniumLux, 2024-01-06:
Hi, thanks for the response.

Dsconnecting the board from Thonny didn't come to my mind but I tried it just now and the result is the same. I flashed the LED code from above to the board from Thonny succesfully, then disconnected the board in Thonny like you said, closed Thonny as well, tried to run the code in PyCharm (with MicroPython setup and the board detected) but the result is the same.

Is it really possible that there is a problem with MicroPython in PyCharm? I know I could just give up and use Thonny since it works well but it's really bothering me and since I will have a lot of files, PyCharm will be more suitable for my situation.

Here you can see a full screenshot

[image]

Thanks :)

gabemorris12, 2024-01-06:
@somniumLux According to your screenshot, there is nothing wrong with PyCharm. It successfully flashed the script to the board. The only thing micropython in PyCharm can't do is run the REPL console. My hope is that one of the Jetbrains developers are working on it.

If you didn't see the pico flashing the light, then it's probably because of an error. Trying changing '"LED"' to '25' on line 4. The string only works for the pico w (I think) and you probably have the normal pico. Regardless, you should be able to see any errors in the script when you run it through Thonney.

somniumLux, 2024-01-06:
You are right, 25 should be used on the original Raspberry Pi Pico board but I'm using the Pico W and the code works in Thonny without any errors and the LED works. And yes, there are no errors in PyCharm and it says that the upload is done but the LED is not flashing.

I don't have any other ideas other than that there is an issue with PyCharm here. I also conntacted their support so maybe I get a useful answer.

Thank you :)

britesc, 2024-03-29:
See
[https://github.com/JetBrains/intellij-micropython/issues/275](url)