JetBrains/intellij-micropython #273: PyCharm, micropython module, does not print any result (output)

renejeanmercier, 2023-12-20:
Hello,

I just installed and ran the following micropython program on and ESP32 :

import machine
import time
pin = machine.Pin(2, machine.Pin.OUT)
while True:
    pin.value(0)
    print("LED ON");
    time.sleep(1)
    pin.value(1)
    print("LED OFF");
    time.sleep(1)

Everything upload fine. It runs fine as well (I checked the output with puTTY and Thonny).

But using PyCharm I don't get any printed out on the screen. 

Is there something I do wrong or I am missing?  Is it something you can please verify?

I am running debian : 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Thank you so much.

Regards,

Rene-Jean

gabemorris12, 2023-12-20:
You have to run it from the REPL console. I believe this feature is temporarily down. You can only flash scripts. See issue #270.

renejeanmercier, 2023-12-20:
Hello Gade,

Sad news. When I click on the issue number you provided, it says that it
was working with a previous version.

Where can I access this version?

Thank you.

Regard,

Rene-Jean
====================================

zerovijay, 2023-12-20:
Pycharm only support ESP8266, I thing this is compatible issue!
if you are using ESP8266 make sure  you enable the Board support
[image]
Sorry i don't have ESP8266 to recreate issue! 
Note: MicroPython REPL not working on New update so wait for next update or use older pycharm version 2023.2.5 it's working just fine.

zerovijay, 2023-12-20:
This is temporary issue. maybe it's fix on next update!

renejeanmercier, 2023-12-20:
Hi Vijay. Where will I find the version you mentioned in your email.

Please give me the hyperlink.

Regards.

René-Jean .

zerovijay, 2023-12-21:
> Hi Vijay. Where will I find the version you mentioned in your email. Please give me the hyperlink. Regards. René-Jean . Le mer. 20 déc. 2023 à 06:27, Vijay ***@***.***> a écrit :
> […](#)

Use this link to get pycharm all version [The link:](https://www.jetbrains.com/pycharm/download/other.html)

renejeanmercier, 2023-12-22:
Hi Vijay,

Thank you. for your reply.

Hoping that it will be fixed in a short while.

You might want to inform me when it will be fixed.

Kind regards,

Rene-Jean
================================

elmot, 2024-07-24:
Probably duplicate to #283

closed 2024-08-29 by elmot