JetBrains/intellij-micropython #274: MicroPython REPL not working!

zerovijay, 2023-12-21:
Old issue: The REPL not fixed this is 3rd update still not fixed

[image]

New issue: Black format failed to install

[image]

PyCharm 2023.3.2 (Community Edition)
Build #PC-233.13135.95, built on December 19, 2023
Runtime version: 17.0.9+7-b1087.9 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.
Linux 6.6.7-200.fc39.x86_64
GC: G1 Young Generation, G1 Old Generation
Memory: 2048M
Cores: 2
Registry:
  ide.experimental.ui=true
Non-Bundled Plugins:
  intellij-micropython (1.4.0)
Current Desktop: GNOME

palko444, 2023-12-24:
Hello,
I have same issue with REPL with PyCharm professional

Build #PY-233.13135.95, built on December 19, 2023
Runtime version: 17.0.9+7-b1087.9 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.

OS: Fedora 39, KDE,
kernel: 6.6.6-200.fc39.x86_64
MEM; 64GB
CPU: AMD Ryzen 7 5800X 8-Core Processor
Non-Bundled Plugins:
intellij-micropython (1.4.0)

Thank you

asdaraujo, 2024-01-05:
Same here, but I found a workaround for the time being.

1. I created a folder in my project (called it ".repl") and copied the [microrepl.py](https://github.com/JetBrains/intellij-micropython/blob/master/scripts/microrepl.py) into it.
2. When I need to use the REPL, I open a Terminal in PyCharm and type in the following command:
```
python .repl/microrepl.py /dev/tty.usbmodem142201
```
Where `/dev/tty.usbmodem142201` is the Raspberry Pico device on my computer.

And voilá!
[image]

closed 2024-07-13 by lancelote