JetBrains/intellij-micropython #275: MicroPython REPL doesn't work with newest Pycharm

MATTYGILO, 2023-12-24:
Version 2023.3
https://www.jetbrains.com/pycharm/download/other.html

ianc1215, 2024-01-20:
Can confirm it also doesn't work with the patch version 2023.3.2.

asdaraujo, 2024-01-20:
@ianc1215 please see [this](https://github.com/JetBrains/intellij-micropython/issues/274#issuecomment-1877979780) for a workaround until there's a fix

ianc1215, 2024-01-20:
How about for Windows 11? This solution appears to be for Linux.

britesc, 2024-03-29:
Bump
Can confirm VSCode and Thonny work perfectly as does mpremote.
Still not working in PyCharm.

albertogomcas, 2024-05-14:
@ianc1215 Instead of dev/tty/... simply use the windows com port of your device

`python .\.repl\microrepl.py com11`

walker2398, 2024-05-15:
It's now 2024.1,and my REPL still doesn't work.

closed 2024-07-13 by lancelote