JetBrains/intellij-micropython #277: Arduino Nano RP2040 Connect not detected

DanieleZiz, 2024-01-13:
Hello!
I'm trying to connect Arduino Nano RP2040 Connect as a Raspberry Pico platform with the plugin on a Windows computer, but it can't be detected, even specifying the correct port (which in my case, was COM4).
The REPL runs flawlessly on COM4 using Arduino Lab for Micropython, but i'd like to use Pycharm for versioning purposes.

pmoses, 2024-01-13:
I have the same issue on RPi Pico.
Solution: Running Thonny, it detects COM port flawlessly.
Close Thonny, add the right com port to the Settings -> Languages & Frameworks -> MicroPython: Device Path: COM6 

[image]

DanieleZiz, 2024-01-13:
But i can see the RP2040 is connected to port COM4 from device manager; the problem is that even writing COM4 in the Device path won't work for me

britesc, 2024-03-29:
See
[https://github.com/JetBrains/intellij-micropython/issues/275](url)

closed 2024-11-11 by elmot