JetBrains/intellij-micropython #283: Micropython REPL not working still!!

britesc, 2024-03-29:
See
[https://github.com/JetBrains/intellij-micropython/issues/275](url)
I cannot understand why the Micropython REPL issue is still unaddressed.
Now that I have it working flawlessly in both VSCode and Thonny I don't know if I need to bother with PyCharm anymore. Shame was just beginning to like it.

britesc, 2024-04-01:
@elmot 
As requested my configuration.

    1.    Linux Leopard 6.5.0-26-generic #26~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Mar 12 10:22:43 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
    2.    MicroPython v1.22.2 on 2024-02-22; Raspberry Pi Pico W with RP2040
    3.    PyCharm 2023.3.5 (Community Edition) Build #PC-233.15026.15, built on March 21, 2024 Runtime version: 17.0.10+1-b1087.23 amd64
    4.    Current Desktop: ubuntu:GNOME

The forum is flooded with people commenting that MicroPython no longer works with PyCharm. I am not an isolated case. So if you have am installation that works, then please post a HOWTO: here for the rest of us unfortunates.

Kind regards,

jB

elmot, 2024-04-02:
Micropython plugin version is not provided in your list. Any diagnostics is not possible until we don't know that.
It's here:
[image]

britesc, 2024-04-02:
@elmot 
Sorry thought you wanted the REPL details at 2. above. Didn't know there was a diference.
Thanks and kind regards,
jB 

[image]

lancelote, 2024-04-02:
@britesc 

> not working

Could you please provide the details? Is there an error of some kind? What do you see in the REPL tool window?

> The forum is flooded with people commenting that MicroPython no longer works with PyCharm. I am not an isolated case

The original problem was fixed in the latest versions

momer, 2024-04-25:
I'm in the same boat as @kyledross ! 

Details
- PyCharm 
   - Version: `2024.1`
   - Build: `PY-241.14494.241, March 28, 2024`
   - Runtime version: `17.0.10+8-b1207.12 amd64`
   - VM: `OpenJDK 64-bit server vm by JetBrains s.r.o`
- MicroPython plugin @ `1.4.0`

jamescrace, 2024-05-20:
2020 M1 Air running Sonoma 14.4.1
PyCharm 2024.1 (Professional Edition)
Build #PY-241.15989.155, built on April 29, 2024
MicroPython 1.4.0

same as @kyledross, presented with a blinking cursor.

nicolascerna, 2024-05-21:
Same here:

macOS 14.4.1 - Apple M1 Pro (Apple Silicon)
PyCharm 2024.1.1 (Professional Edition) - Build #PY-241.15989.155
MicroPython Plugin v1.4.0

Everything else working fine in Arduino for MicroPython and VSCode.

philliphodges, 2024-05-21:
Same issue here..!

Dell Precision 7670
Windows 11 Pro - 23H2 - 22631.3593

PyCharm 2024.1.1 (Professional Edition)
Build #PY-241.15989.155, built on April 29, 2024
Runtime version: 17.0.10+1-b1207.14 amd64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.

This is the error thrown when trying to push to device...

```
2024-05-21 12:05:32,959 [ 598552] SEVERE - #c.i.o.a.i.FlushQueue - 'com.pty4j.PtyProcess com.jetbrains.micropython.repl.MicroPythonReplManager$startREPL$terminalRunner$1.createProcess(org.jetbrains.plugins.terminal.TerminalProcessOptions, com.intellij.terminal.JBTerminalWidget)'
java.lang.NoSuchMethodError: 'com.pty4j.PtyProcess com.jetbrains.micropython.repl.MicroPythonReplManager$startREPL$terminalRunner$1.createProcess(org.jetbrains.plugins.terminal.TerminalProcessOptions, com.intellij.terminal.JBTerminalWidget)'
	at com.jetbrains.micropython.repl.MicroPythonReplManager.startREPL(MicroPythonReplManager.kt:64)
	at com.jetbrains.micropython.run.MicroPythonRunConfiguration$getState$1.invoke$lambda-0(MicroPythonRunConfiguration.kt:90)
	at com.intellij.util.concurrency.ContextRunnable.run(ContextRunnable.java:27)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:209)
	at com.intellij.openapi.application.TransactionGuardImpl.access$100(TransactionGuardImpl.java:22)
	at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:191)
	at com.intellij.openapi.application.impl.RwLockHolder.runIntendedWriteActionOnCurrentThread(RwLockHolder.kt:204)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:830)
	at com.intellij.openapi.application.impl.ApplicationImpl$2.run(ApplicationImpl.java:419)
	at com.intellij.openapi.application.impl.RwLockHolder.runWithEnabledImplicitRead(RwLockHolder.kt:138)
	at com.intellij.openapi.application.impl.RwLockHolder.runWithImplicitRead(RwLockHolder.kt:129)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWithImplicitRead(ApplicationImpl.java:1152)
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:81)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:123)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:43)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:792)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:739)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:733)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:761)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.kt:699)
	at com.intellij.ide.IdeEventQueue._dispatchEvent$lambda$12(IdeEventQueue.kt:593)
	at com.intellij.openapi.application.impl.RwLockHolder.runWithoutImplicitRead(RwLockHolder.kt:105)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.kt:593)
	at com.intellij.ide.IdeEventQueue.access$_dispatchEvent(IdeEventQueue.kt:77)
	at com.intellij.ide.IdeEventQueue$dispatchEvent$processEventRunnable$1$1$1.compute(IdeEventQueue.kt:362)
	at com.intellij.ide.IdeEventQueue$dispatchEvent$processEventRunnable$1$1$1.compute(IdeEventQueue.kt:361)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:843)
	at com.intellij.ide.IdeEventQueue$dispatchEvent$processEventRunnable$1$1.invoke(IdeEventQueue.kt:361)
	at com.intellij.ide.IdeEventQueue$dispatchEvent$processEventRunnable$1$1.invoke(IdeEventQueue.kt:356)
	at com.intellij.ide.IdeEventQueueKt.performActivity$lambda$1(IdeEventQueue.kt:1022)
	at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:106)
	at com.intellij.ide.IdeEventQueueKt.performActivity(IdeEventQueue.kt:1022)
	at com.intellij.ide.IdeEventQueue.dispatchEvent$lambda$7(IdeEventQueue.kt:356)
	at com.intellij.openapi.application.impl.RwLockHolder.runIntendedWriteActionOnCurrentThread(RwLockHolder.kt:209)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:830)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.kt:398)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:207)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:128)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:117)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:105)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:92)
2024-05-21 12:05:32,960 [ 598553] SEVERE - #c.i.o.a.i.FlushQueue - PyCharm 2024.1.1  Build #PY-241.15989.155
2024-05-21 12:05:32,960 [ 598553] SEVERE - #c.i.o.a.i.FlushQueue - JDK: 17.0.10; VM: OpenJDK 64-Bit Server VM; Vendor: JetBrains s.r.o.
2024-05-21 12:05:32,960 [ 598553] SEVERE - #c.i.o.a.i.FlushQueue - OS: Windows 11
2024-05-21 12:05:32,960 [ 598553] SEVERE - #c.i.o.a.i.FlushQueue - Plugin to blame: MicroPython version: 1.4.0
2024-05-21 12:05:32,960 [ 598553] SEVERE - #c.i.o.a.i.FlushQueue - Last Action: Run
```

kcunko, 2024-05-31:
Same here, Windows 11 WSL2 Ubuntu Linux 5.15.146.1-microsoft-standard-WSL2
PyCharm 2024.1.2 (Professional Edition) #PY-241.17011.127, built on May 28, 2024
Plugin version 1.4.0 (latest)
REPL console not working, blinking cursor.

lenineto, 2024-06-19:
Same issue for me. Nothing is printed on console. 
Running PyCharm 2024.1.3 (Professional Edition) Build #PY-241.17890.14 on MacOS Sonoma 14.5 (23F79).

AinoSpring, 2024-06-26:
Same here, on
PyCharm 2024.1.4 (Professional Edition)
Build #PY-241.18034.82, built on June 24, 2024
Linux 6.8.0-35-generic
Registry:
  ide.experimental.ui=true
Non-Bundled Plugins:
  IdeaVIM (2.12.0)
  intellij-micropython (1.4.0)
Current Desktop: ubuntu:GNOME

elmot, 2024-07-23:
Please re-verify using updated plugin version

kyledross, 2024-07-23:
@elmot Yes, it is working now.  Thank you!

lenineto, 2024-07-24:
@elmot mine still doesn't work. Same issue as before. Am I missing someting?
I'm running PyCharm 2024.1.4 (Professional Edition) Build #PY-241.18034.82, built on June 24, 2024
and Micropython plugin 1.4.4-2024.1 from July 13, 2024.
Thanks!

elmot, 2024-07-24:
@lenineto 
What if you click repl restart button?

lenineto, 2024-07-24:
Hi @elmot, thanks for your quick reply.
No luck with repl restart. Clicking restart simply interrupts the program (and generates an error message). I'm running a test script that changes LED color and prints the current state to console. When running on Thonny I can match the messages with the LED color. When running on PyCharm, I get the expected LED behaviour but as before, nothing printed on console.

Luky557, 2024-07-24:
Does it talk about a Keyboardinterrupt error? If so, try the RESET button if your device has one

lenineto, 2024-07-24:
Hey @Luky557, yes it does mention a KeyboardInterrupt error, but only if I click the repl restart button. Otherwise the program runs fine, just nothing printed on console. I'm using a Pico W, so I don't have a reset button. Cheers!

Luky557, 2024-07-24:
https://electrocredible.com/how-to-reset-raspberry-pi-pico-w/#:~:text=Connecting%20a%20push%2Dbutton%20between,Pi%20Pico%20and%20Pico%20W.

You could try using one of the pins too, it's been quite some time since I have done anything with the Pico W, but what you want to do if you are getting Keyboard Interrupts is to hardware reset the Pico.

lenineto, 2024-07-24:
I appreciate the tip, @Luky557 but I'm stuck on the same problem. If I connect Pin30 to ground, the pico does reset, I can once again see the LEDs working as expected after the reset, but still, nothing on console.

Luky557, 2024-07-24:
I just want to double check. You have the configuration of micropython set to flash, you have toggled open MicroPython REPL on success and before launch task is Stop MicroPython REPL?

After that, for me with an ESP32 it runs the REPL and quickly falls into Keyboardinterrupt error, at that point I can just hardware restart the ESP32 and without even touching PyCharm console output starts working as expected. I acknowledge that it might now work like this for you, I am just ensuring you are doing what I am to try and get it working.

Luky557, 2024-07-24:
Please do the following, if it doesn't work, please attach screenshots of what the config I suggest looks like for you and what the console output looks like. Navigate to the flash main.py clkick on it and select edit configurations
[image]

Make sure your settings look similar to mine. The REPL on success, stop micropython before launch and activate tool window should be configured as they are for me.

Once that is done, flash, let it run and after it does you should see a REPL open and a keyboardinterrupt printed. 

This is when you hardware reset the device, the console should receive more output and start working just fine.

lenineto, 2024-07-24:
Hi @Luky557 
My settings are exactly the same as yours
[image]

This is the REPL console output right after Run 'Flash main.py'
[image]
You can see the KeyboardInterrupt message there.

After the hardware reset as suggested, this is what we get
[image]

Just to make it clear, there are no errors in main.py, as the message suggest. I can verify the program is running fine, by looking at the LED.

elmot, 2024-07-24:
>You can see the KeyboardInterrupt message there.

Try Ctrl-D in REPL window

lenineto, 2024-07-24:
@elmot that's it! Ctrl+D did the trick. No need to hard reset or anything. Just Run Flash main.py, hit CTRL+D and it works.
Looks like we need the force the soft reboot to get REPL working, I thought the "Stop Micropything REPL" on the config would handle that, and I never really paid attention to that CTRL+D message.

Thanks a bunch!

Thank you for your attention and support as well, @Luky557. Much appreciated.

closed 2024-07-27 by lancelote

lancelote, 2024-07-27:
1.4.5 plugin version with the fix is released. Please give it a try and share your feedback.

lenineto, 2024-08-05:
Hi @elmot and @lancelote 
The new plugin version fixes the issue and the REPL starts working immediately after flashing a file, however I noticed two new bugs:
1. no matter what file you flash, it will always run main.py (that's the output on REPL)
2. hitting CTRL - D for reset no longer works

Is there anything else that changed and I'm unware?

Cheers

Luky557, 2024-08-05:
main.py is run by default by design. MicroPython automatically looks for files boot.py and main.py those are ran, others have to be ran from them. I can't say anything about the CTRL D, never need to use it.

PR #313 (2024-07-27): Don't interrupt program execution when REPL is started right after upload.

Fixes #283
commit 0fe90222 (2024-07-27): Don't interrupt program execution when REPL is started right after upload.

Fixes #283