JetBrains/intellij-micropython #292: Micro Python REPL无法正常使用

soulmate543748, 2024-07-07:
中：我使用pycharm2024.1连接ESP32-WROOM-32，并烧录程序
EN：I use pycharm 2024.1 connect ESP32-WROOM-32 and burn the program
[image]
中：程序可以正常写入，但REPL无法显示，或显示SESSION HAS BEEN INTERRUPTED
EN：The program can be written normally, but the REPL cannot be displayed, or SESSION HAS BEEN INTERRUPTED is displayed.
[image]
[image]
中：非常抱歉，这是我第一次在GitHub上说明问题，英语是使用翻译软件生成的，有表达不清楚的地方，还请见谅
EN：I'm very sorry, this is my first time explaining the problem on GitHub. English is generated using translation software. Please forgive me for some unclear expressions.

UF4OVER, 2024-07-07:
解决了吗，一样的问题，我都想自己写一个了

soulmate543748, 2024-07-10:
*问题解决*
发现是因为在烧录的时候，将ESP32文件夹烧录入，所以使用`import LED`，无法识别到文件
可以先在`MicroPython REPL`中使用
```python
import os
os.listdir()
```
查看有哪些文件在硬件内，再根据所要烧录的文件位置修改，如我在上图中出现的错误，修改后应该是
```python
import ESP32.LED
```
注意：一定要是`LED`，而不是`LED.py`

closed 2024-07-10 by soulmate543748