JetBrains/intellij-micropython #317: print outputs not shown in REPL

Omid888, 2024-08-02:
I know that multiple issues is already open about this manner right now, but this bug is so crucial that has stopped me from using pycharm.
Micropython -by its nature- does not give developers any debugging tools and not being able to print is a deal breaker.
So please fix this.
if any other information is needed please inform me to provide.

pycharm version:
PyCharm 2024.1.4 (Community Edition)
Build #PC-241.18034.82, built on June 24, 2024
Runtime version: 17.0.11+1-b1207.24 x86_64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o.

micropython version:
1.23.0

plugin version:
1.4.5-2024.1

OS: macos Sonoma 14.5

Luky557, 2024-08-02:
Prints work just fine for me and so does auto device detection, could you tell me what micro controller you're using?

Omid888, 2024-08-03:
@Luky557 

My device is raspberrypi pico.

It seems that whenever the code is correct, the print statement works and shows up in micropython REPL console.
However if there is a syntax error in the code, not only the values are not printed, the error does not print on screen either. You have to manually open `Tool window > micropython` and click restart icon to gain the REPL prompt and type `import main.py` to print the errors in the code.

Besides the runtime errors not being shown, why syntax checking is not done before uploading the code? (like Thonny & vscode)

lukaskremla, 2024-08-13:
Sorry for late response I forgot about commenting here. I think print works just fine, but you have to get used to the brokenness of REPL right now.

Unless you refrain from updating to the latest PyCharm, you will have a broken REPL altogether until the devs fix it, if you are still on the previous version where you can get REPL to open you need to get used to the following:

REPL struggles to automatically launch after flashing, the problem was worse before, a recent update made it less bad, but it is still very inconsistent. 

I recommend setting the flash configuration Path to your project folder, so you upload everything in it, if you want to avoid uploading specific folders you can exclude them by right clicking on them and navigating to "mark directory as", you can't exclude individual files.

Aside of the path ensure to also tick "Open MicroPython REPL on success".

Now, when you flash through this configuration REPL should open after flashing is done. Chances are you get a Keyboardinterrupt, working code or... well... nothing at all. What works for me on the ESP32 S2 is using one of the shortcuts which get printed in the intro message when REPL opens to restart the board or alternatively pressing the reset button on my board. 

The pico might be missing that, another user on this github in a different issue said that they tried restarting the device through the pin on the board, but that did not work and what got it working for them was the shortcut. 

So I can get my ESP32 S2 working and printing through resetting like this, you have to try what works for your Pico, fingers crossed that the big update to this extension is coming soon to fix this and a plethora of other issues I am aware of, in the mean time this is what we have.

Omid888, 2024-08-14:
1)
> Unless you refrain from updating to the latest PyCharm
I do have the latest version of pycharm. (as noted above)

2)
Freezed REPL is not a problem for me. My problem is that REPL does not show the error massages.

3)
> the big update to this extension is coming soon to fix this
I think I will wait for the update, and keep using vscode for now.
I appreciate the information you provided here. Thank you.