JetBrains/intellij-micropython #319: Pycharm false alarm: Unresolved attribute reference 'toggle' for class 'Pin'

Omid888, 2024-08-02:
Pycharm throws a warning for this working code:

```
from machine import Pin
from time import sleep

some_led = Pin(25, Pin.OUT)

while True:
    some_led.toggle()
    sleep(1)
```

Although the code works correctly on the microcontroller, pycharm could not understand the toggle function and shows this warning:
`Unresolved attribute reference 'toggle' for class 'Pin' `

Since other IDEs (vscode & thonny) does not have this problem, I reached out to report it.

As you have probably guessed, pycharm's auto-completion does not work for this function, but it works for other functions.

lancelote, 2024-08-03:
> the code works correctly on the microcontroller

Would you kindly specify the board?

Omid888, 2024-08-03:
@lancelote 

raspberrypi pico
micropython version: 1.23.0

Omid888, 2024-08-03:
`import onewire` has this problem too.

I checked the micropython library in `External Libraries` section. It does not contain the `onewire.pyi` file.

lukaskremla, 2024-08-13:
There is a similar issue with const, which has to be imported as `from micropython import const` to make PyCharm errors go away, works perfectly fine on the micro controller regardless of the import.

Omid888, 2024-08-16:
@Josverl Do you have any suggestions?

Josverl, 2024-08-18:
I'm assuming that you use the micropython-stubs for this, and followed [the pycharm config for it](https://micropython-stubs.readthedocs.io/en/main/24_pycharm.html)

`const` is indeed a builtin that does not need an import on micropython, and for pylance/py right this can be configured through a `__builtin__.pyi` stub.

I have not yet found a way to configure the same in pycharm