JetBrains/intellij-micropython #321: Latest PyCharm broke REPL again

lukaskremla, 2024-08-13:
As the title suggest, the REPL is stuck on a blank console showing no output with the latest PyCharm version on MacOS Sonoma 14.6.1 running PyCharm Build #PY-242.20224.347, built on August 9, 2024.

The board I am using is ESP32 S2, but judging from the previous REPL issues, it will be broken for all boards.

peterzdeb, 2024-08-16:
Same problem on Ubuntu after last upgrade to PyCharm 2024.2 (Professional Edition). I was trying to uninstall and install the MicroPython plugin again. Nothing helps to make the REPL working, but I can upload the source code to the device.

I'm using Raspberry Pico.

lukaskremla, 2024-08-16:
The plugin is broken, I understand your reflex to trying to reinstall it or fix it on your end, but that is simply not possible, it has been this way many times before and it won't work with the latest PyCharm version without an intervention form the devs. 

In the mean time, I recommend you to downgrade to PyCharm version 2024.1.6 on which this plugin still works. That is what I did.

peterzdeb, 2024-08-16:
I'm using `minicom` so it's not a big problem but the life could be much better with REPL in that plugin.

Here is an error I've found in my PyCharm logs:
```
2024-08-16 10:53:19,695 [ 820197] SEVERE - #c.i.o.o.e.ConfigurableWrapper - No display name specified in plugin descriptor XML file for configurable com.jetbrains.micropython.settings.MicroPythonProjectConfigurable;
specify it using 'displayName' or 'key' attribute to avoid necessity to load the configurable class when Settings dialog is opened [Plugin: intellij-micropython]
com.intellij.diagnostic.PluginException: No display name specified in plugin descriptor XML file for configurable com.jetbrains.micropython.settings.MicroPythonProjectConfigurable;
specify it using 'displayName' or 'key' attribute to avoid necessity to load the configurable class when Settings dialog is opened [Plugin: intellij-micropython]
	at com.intellij.openapi.options.ex.ConfigurableWrapper.getDisplayName(ConfigurableWrapper.java:158)
	at com.intellij.ide.util.gotoByName.GotoActionModel.lambda$new$0(GotoActionModel.java:87)
	at com.intellij.util.concurrency.SynchronizedClearableLazy._get_value_$lambda$1$lambda$0(SynchronizedClearableLazy.kt:41)
	at java.base/java.util.concurrent.atomic.AtomicReference.updateAndGet(AtomicReference.java:210)
	at com.intellij.util.concurrency.SynchronizedClearableLazy.getValue(SynchronizedClearableLazy.kt:40)
	at com.intellij.util.concurrency.SynchronizedClearableLazy.get(SynchronizedClearableLazy.kt:28)
	at com.intellij.ide.util.gotoByName.GotoActionModel.getConfigurablesNames(GotoActionModel.java:391)
	at com.intellij.ide.util.gotoByName.ActionAsyncProvider.optionsFlow(ActionAsyncProvider.kt:305)
	at com.intellij.ide.util.gotoByName.ActionAsyncProvider.access$optionsFlow(ActionAsyncProvider.kt:35)
	at com.intellij.ide.util.gotoByName.ActionAsyncProvider$filterElements$2$channel$1.invokeSuspend(ActionAsyncProvider.kt:89)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:104)
	at kotlinx.coroutines.internal.LimitedDispatcher$Worker.run(LimitedDispatcher.kt:111)
	at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:99)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:608)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:873)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:763)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:750)
```

Radiocode1969, 2024-08-27:
Same issue on Windows 11 Home with Pycharm Professional 2024.2.0.1. REPL is blank.

Uploading and running code works fine, but I cannot get the REPL to work, nor can I get output from print() on the Pico in the IDE. All works fine with Thonny.

I will try downgrading to 2024.1.6 and see if that works.

gabemorris12, 2024-08-27:
I'm getting the same thing, and it's also the same with the Mac.

elmot, 2024-08-29:
We've fixed a mess with versions.
Please try to upgrade the plugin now.

closed 2024-08-29 by elmot