JetBrains/intellij-micropython #329: Broken in PyCharm 2024.2.4

gabemorris12, 2024-11-02:
The REPL console does not properly boot up, and I can no longer flash a script with the latest update. Flashing the script gets hung up, and the console will not stop the program. It will show the output from the pico, but it will never halt the program for me to actually run single commands. This was all done on a Mac.

elmot, 2024-11-03:
@gabemorris12 how about to test a new approach?
https://github.com/JetBrains/intellij-micropython/discussions/330

jonhp, 2024-12-12:
I have this same problem. I hope there will be a fix soon. Thanks, Jon

rovrevik, 2024-12-21:
same broken compatibility

gabemorris12, 2024-12-21:
@jonhp @rovrevik Follow #330 issue above. You need to install the revamped version. It's much better!