JetBrains/intellij-micropython #341: NOVA - Properly handling opening files on the device and REPL auto clear

lukaskremla, 2024-11-17:
In the latest pre-alpha 12 opening a file on the device file system with `Auto Clear REPL` enabled doesn't clear the REPL terminal like it did in the previous versions, but each time a file is transferred via serial communication, so that it can be accessed, the intro message gets printed:

```
MicroPython v1.24.0 on 2024-10-25; Generic ESP32S2 module with ESP32S2
Type "help()
```

Now, if I am not mistaken, the file transfer halts the program running on the device, and the only way to make it run again is to flash a new file or send the RESET command either via a GUI action or via a shortcut, all of these options should correctly trigger the `Auto Clear REPL` functionality. 

Would it be possible to also suppress the intro message specifically for the file transfers which happen to facilitate the "Open File" functionality?

If so, it seems like the last step necessary to polish the interaction between the file opening and REPL auto clearing functionalities.

elmot, 2024-11-17:
Thanks for the report.
After some discussion at mpy discord, I decided to freeze any activities on communication improvements.
There is a big chance that if https://github.com/micropython/micropython/pull/14141 is merged, I'll reevaluate if mpremote may be used in the plugin. 

Until that evaluation I see no sense to improve a working(but not perfect) module. Of course little changes and bugs to be fixed anyway.

Anyway, either my approach or mpremote - most probably any file operation breaks a running program execution silently, so if a visibility of that fact is required - probably better to let me how it should look like

lukaskremla, 2024-11-17:
> Thanks for the report. After some discussion at mpy discord, I decided to freeze any activities on communication improvements. There is a big chance that if [micropython/micropython#14141](https://github.com/micropython/micropython/pull/14141) is merged, I'll reevaluate if mpremote may be used in the plugin.
> 
> Until that evaluation I see no sense to improve a working(but not perfect) module. Of course little changes and bugs to be fixed anyway.
> 
> Anyway, either my approach or mpremote - most probably any file operation breaks a running program execution silently, so if a visibility of that fact is required - probably better to let me how it should look like

Well there are two ways.

One is the way it is now, but suppressing the 

```
MicroPython v1.24.0 on 2024-10-25; Generic ESP32S2 module with ESP32S2
Type "help()
```

Message, because it happens for every file you try to open, which is unnecessary. And maybe adding a singular message, that won't be spammed for every newly opened file. Something like "Device program stopped to open a file" (the exact message can be adjusted). 

The other option, which is the way thonny goes. Is that it doesn't allow you to open a file located on the file system of the device as long as there's code running on the device, it will just keep throwing an error. Now, it does this even for file transfers, uploads etc. And from my experience working with Thonny it can get quite annoying.

Maybe if you were to stay with the current solution a toggle between the two described behavior would be nice in the future.

However, knowing what you said regarding potential move to mpremote - in order to make the plugin work in a sufficient way now, I would recommend option 1.