JetBrains/intellij-micropython #359: compatibility problem

mitsh, 2025-02-13:
Hello Folks!

Currently, MicroPython doesn't work with new version of JetBrains products. When will you release to support new Jetbrains editors?

Cheers!

jorgechacblogspot, 2025-02-16:
**The micropython plugin is not compatible with pycharm version 2024 3.3. Please update it.**

waanito, 2025-02-16:
As far as I know support for micropython is 3rd party…

jorgechacblogspot, 2025-02-16:
https://plugins.jetbrains.com/plugin/9777-micropython

waanito if you read a little you would know that the micropython plugin for Pycharm is developed by JetBrains s.r.o. who are the ones who manage this GitHub.

waanito, 2025-02-16:
Thanks. I stand corrected.
My confusion is from this post
https://github.com/JetBrains/intellij-micropython/issues/123
where jjjdm wrote on Nov 2, 2020
"Keep in mind this is not an Officially Supported Feature of PyCharm but rather just a JetBrains employee that was nice enough to create the micropython plugin."

lukaskremla, 2025-02-16:
Unfortunately the original author is no longer active in this project. There were some recent attempts to resurrect this project, but those have been inactive for a while now as well. However, you can find other MicroPython plugins in the plugin marketplace of the latest PyCharm release.

mitsh, 2025-02-17:
thanks for the explanation @lukaskremla thats really sad. I found a fork of this project https://github.com/lukaskremla/micropython-tools-jetbrains its good for now.

jorgechacblogspot, 2025-02-20:
The current micropython plug-in for pycharm does not work the same as the one from JetBrains s.r.o. I will personally stop using pycharm and migrate to Thonny.