JetBrains/intellij-micropython #360: ESP32 Wroom DevKit4

anyone14959, 2025-04-25:
Hi there,
since v0.5x it's not possible anymore to establish a connection, neither by COM nor by webrepl....

Nedded to downgrade

best regards...

lukaskremla, 2025-04-25:
You've mentioned version 0.5.x and WebREPL, given that this intellij micropython plugin is currently not even available in the plugin marketplace - perhaps you've meant to go here? - https://github.com/lukaskremla/micropython-tools-jetbrains/issues

anyone14959, 2025-04-25:
Yes, you are right. Can you move this issue or shall I create a new issue at the right place?

lukaskremla, 2025-04-25:
> Yes, you are right. Can you move this issue or shall I create a new issue at the right place?

I'm not affiliated with this repository. it'd be best if you created a new bug report in the correct repository and filled out the bug report template.

closed 2025-04-26 by anyone14959